victronenergy/dbus_modbustcp #8: ESS control loop setpoint 2703 vs ESS control loop setpoint 2700

warwickchapman, 2020-10-05:
There appears to be a gap in the ranges and scafactors of 2700 and 2703.

Remark: `ESS Mode 2 - Setpoint for the ESS control-loop in the CCGX. The control-loop will increase/decrease the Multi charge/discharge power to get the grid L1 reading at this setpoint`
Range: `-32668 to 32668`
Scalefactor: `1`

Remark: `ESS Mode 2 – Same as 2700, but with a different scale factor. Meant for values larger than 65kW.`
Range: `-3266800 to 3266800`
Scalefactor: `0.01`

Questions:

1. Given that we can set negative setpoints, and there is a reason to do so, *and* given the `Scalefactor` of `1` for `2700`, it only allows setting a maximum 32668W / ~32kW.

2. The remark in `2703` states that this register is intended for system which require a setpoint "larger than 65kW".  Where does the 65 come from?  We could understand it is there were no negative setpoints for `2700`, but given that there are, why 65kW?

3. Given the `Scalefactor` of `0.01` for `2703`, doesn't that render `2703` functionally equivalent to `2700`?

`3266800 * 0.01 = 32 668`

closed 2020-10-05 by izak

warwickchapman, 2020-10-06:
Hey Izak

Thanks for the changes.  They resolve only the "gap" portion of my Github issue but I'm not yet following on the scalefactor.

May I ask for clarification:

Further to your response in #7, I seem to be misunderstanding `Scalefactor` because with `2700` having `1` and `2703` having `0.01`, it would seem to me they are functionally equivalent.

In #7 you explained that a `Scalefactor` of `10` means "[i]t allows you to set the minsoc in tenths of a percentage, which is even finer granularity than on the graphical interface."

If that is the case, and it means the same as it does in `270x`, then are we talking `0.01`'s of a Watt.  Perhaps the unit for 2703 supposed to be `kW`?

izak, 2020-10-06:
The scale factor essentially tells you how many places to move the decimal point, which you need to do since modbus inherently use combinations of 16-bit integer values.

For a scale factor of 10, it means move it left one place (divide by 10). So the integer value 500 means 50.0.

For a scale factor of 0.01, it means move the decimal point right two places (divide by 0.01 or multiply by 100). So writing 32000 to such a register (close to the upper limit of an int16) would mean 3,200,000W, or 3.2MW.

So 2700 and 2703 does the same thing (functionality equivalent), but 2703 adds two zeroes to the end of whatever you write and allows you to set values larger than 32kW. The thinking behind that is of course that when you work with large power values like that, the nearest 100W is likely good enough.

There are other ways to do it of course, but this is te way that was historically chosen for this document.

If you are interested, you can look at the sunspec documentation. You will see the INTSF notation used there simply states the number of places to move the decimal point, a 1 means multiply by 10 and a -1 (32768) means divide by 10. Same problem, different solution.

warwickchapman, 2020-10-06:
Beautiful.  Very elegant. 

Thank you for taking the time to write that.  We are very greatful!