victronenergy/dbus_modbustcp #9: Incorrect Readout values for: com.victron.solarcharger ;"User Yield" ; Address: 790

pflugj, 2021-06-14:
I am having a problem with the values:
com.victron.solarcharger ; "User Yield" ; Address: 790

System has 2 SmartSolarCharger MPPT 150/85
Connected via VE.Direct Port 1 & 2 
VRM Instance:
258 > UnitID=245
260 > UnitID=243

Value readout according to "Victron Connect" App connected via Bluetooth should be: 
UnitID 245 ~ 9428kWh - Modbus read Value  = 2874.6kWh
UnitID 243 ~11200kWh - Modbus read Value = 5120.3kWh

This is the ONLY value I have a problem with.

Best regards
Jurgen

RyaLynn, 2022-02-24:
Hey, 

I had a similar problem with the solarCharger, even though it was for more values than just the user yield. I think the main problem (at least in my case) was, that the address 777 is not used by Victron Energy, so my counting of the variables has been one place off, when I read all solarCharger registers with one FC3 call. 
Now I think I have the correct values, assumed that it is called 'total yield' in the solar charger tab in the CCGX. I'm not 100% sure if it is the same, but the values in the Demo mode are equal. Can't test it with a real device for now. 

Rya

pflugj, 2022-02-25:
Thanks for the reply.

Unfortunately I do not think this is the problem.
I have posted the same issue here with some screenshots:
https://community.victronenergy.com/questions/58216/venus-gx-modbus-tcp-incorrect-values-for-user-yiel.html
You can see the readout values from register 771 to 791 for both Unit Id's
It works well for both and even though register 777 is not specified it still delivers a value.

But to make sure I checked again today by starting the read out at register 790 
And now it is completely off on both Unit Ids :-(

**Then I realized that I missed something!!**
On to the CCGX Register list:
com.victronenergy.solarcharger | User yield | 790 | uint16 | 10 | **0 to 6553.5** | /Yield/User | no | kWh

**The field is defined to have values from **0-6553.5 kWh** ONLY!!**

.. and both my units are far above that ~15424kWh and 12624kWh as shown in VictronConnect (via Bluetooth)
Unit ID 245 registers has overflowed at least twice!!
Unit ID 243 registers has overflowed at least once!! 
Accounting for the overflows:
Now Unit ID 245 shows 2316.4 >> 15424kWh - 2x 6553.5kWh  = 2317kWh which is about right!
Now Unit ID 243 shows 6060.8 >> 12615kWh - 1x 6553.5 kWh = 6061.5kWh which is also about right!

How to figure out via Modbus how many times an overflow has occurred?? I don´t know

Buy the time I posted this, only the 
Unit ID 243 had a **1x overflow**n and 
Unit ID 245 had **NO** overflow.

BOTH REGISTERS SHOWED THE CORRECT VALUE when considering that overflows did occur!
Meaning I was wrong ;-( 

But I do not bother any longer as I am now using the VENUS MQTT-server  and these values are correct.

Cheers and have a nice day

closed 2022-02-28 by izak