victronenergy/dbus_modbustcp #14: Ability to change AC Input

XBeg9, 2022-09-13:
Based on this discussion, is it possible to change AC Input type using Modbus? https://community.victronenergy.com/questions/146970/generator-and-pv-inverter-on-ac-out.html

kaischmitz, 2023-01-02:
I was also looking for the ability to change the input source type.

I found this works fine via MQTT through the following topics:

W/00e93aac499b/settings/0/Settings/SystemSetup/AcInput1
W/00e93aac499b/settings/0/Settings/SystemSetup/AcInput2

@XBeg9 The register 4520 you mentioned is in the com.victronenergy.multi realm, I can't seem to even read those, tried unit ids 100, 227 and 228.

kaischmitz, 2023-01-02:
I have created a pull request https://github.com/victronenergy/dbus_modbustcp/pull/16.

XBeg9, 2023-02-15:
Victron Team, can somebody look at this? We can celebrate half year for the issue without official response :)