victronenergy/dbus_modbustcp #30: Incorrect unit of measurement for register 1052 in XLSX

sfstar, 2023-05-16:
Hello and thank you for opening the code to the modbus server component of the victron ecosystem.
There seems to be a mismatch between the values documented and actually returned for the dbus path: /ErrorCode

The XLSX states:
[image]

and attributes.csv contains:
'com.victronenergy.pvinverter,/Ac/Power,i,kW,1052,int32,1,W'

The value currently returned by 1052 is wattage based (as matches the attributes.csv file).
However, the XLSX states kW as the correct unit of measurement.
Is the XLSX incorrect or should the value returned be kW based but isn't encoded as such?

For reference the issue on my integrations issue tracker:
https://github.com/sfstar/hass-victron/issues/13

sfstar, 2024-02-27:
@izak could you perhaps give me any insight as to how this discrepancy needs to be addressed/interpreted?

izak, 2024-02-27:
This is already corrected in the xlsx. But thanks for reminding me that we need to update the one on the main website to match the 3.20 release.

closed 2024-02-27 by izak