victronenergy/dbus_modbustcp #31: Incorrect unit of measurement for register 2808

sfstar, 2023-05-16:
Hello and thank you for opening the code to the modbus server component of the victron ecosystem.
There seems to be a mismatch between the values documented and actually returned for the dbus path: /Altitude

The XLSX states:
[image]

and attributes.csv contains:
'com.victronenergy.gps,/Altitude,d,m,2808,int32,10,R'

The value currently returned by 1052 should be in meters.
However, the XLSX states m/s as the correct unit of measurement.
I believe the XLSX should be updated to reflect the altitude as meters but I don't want to diverge from the XLSX without confirming it is wrong.

For reference the issue on my integrations issue tracker:
https://github.com/sfstar/hass-victron/issues/28

askpatrickw, 2023-09-05:
It would be great if someone from the core team could confirm register 1052's units are defined incorrectly as meters per second and should be meters. 🙏

izak, 2023-09-05:
Well clearly the unit is wrong, we need a unit of length, not one of speed! It's corrected in development and will reflect in the official document when we release the next version of Venus. Thank you for reporting this.

closed 2023-09-05 by izak