victronenergy/dbus_modbustcp #32: Non defined modbus slave id's can be picked for devices

sfstar, 2023-05-16:
Hello and thank you for opening the code to the modbus server component of the victron ecosystem.
There seems to be a mismatch between the slave id's picked for devices connected to the GX devices and the documented possible slave id's.

The XLSX states:
[image]

However it seems that other id's might also be selected for the PVinverter devices that are connected to the GX device.
In the issue on my issue tracker a user experencied the GX device picking 34,35,36 as modbus slave id's for his pvinverter devices.

For reference the issue on my integrations issue tracker:
https://github.com/sfstar/hass-victron/issues/36

According to the XLSX data none of these ID's are valid pvinverter slave id's.
Is this by design?
If yes, why are slave id's 35 and 36 not defined in the XLSX?

My integration treats every slave id the same and attempts to retrieve all possible registers for each slave id.
So it doesn't matter to me if the devices don't adhere to the assigned unit id's (although it's still wierd that it diverges from the documentation).
However, I'm not testing id's that aren't defined in the XLSX.
Therefore, I'm also asking if you could provide me with an complete list of possible slave ID's so that i can reflect this in my integration.

sfstar, 2024-03-13:
@izak Would you be able to provide me with an list containing all valid unit id's that the GX device can assign?

izak, 2026-10-08:
This was resolved in Venus 3.60, when we added a note to the mappings tab in the spreadsheet explaining that the slave id is the preferred id, but another will be automatically assigned if the preferred id is already in use.

In Venus 3.70, we made the mapping optional. If the device has a DeviceInstance below or equal to 247, and can be found directly, the mapping is not used.

This largely also answers the question about a list of possible ids that can be assigned: Any ID can be assigned, from 0 to 247. The mapping is communicated via dbus, and displayed on the GUI.

closed 2026-10-08 by izak