victronenergy/dbus_modbustcp #37: User are reporting undocumented solarcharger states

sfstar, 2024-03-13:
Hello,

The victron modbus documentation specifies a list of possible solarcharger states that can be returned by the victron modbus TCP server.
However, users are reporting other states being reported that aren't documented.
https://github.com/sfstar/hass-victron/issues/178

Would it be possible to update the XLSX documentation with the specification of  what state the value 245 represents?

Regards,

Sfstar

wernerhp, 2024-03-14:
`"245": "Off",`
[Source](https://github.com/victronenergy/node-red-contrib-victron/blob/30d49be6d27a4c172aef589ed335111b818d8e54/src/services/services.json#L4402)

wernerhp, 2024-03-14:
Would like to know what `258` is.  It doesn't show up on VRM, but got it on MQTT
`N/system/0/SystemState/State`

[image]
[image]

closed 2026-10-06 by izak