victronenergy/dbus_modbustcp #39: Mismatch in Modbus spreadsheets

gb095666, 2024-03-27:
In the technical area https://www.victronenergy.com/support-and-downloads/technical-information

There are 2 spreadsheets which show Modbus addresses

EVCS-Modbus-TCP-register-list-v2.2 and CCGX-Modbus-TCP-register-list-3.20

There is an error with the description for addresses 3027 and 5026 which both relate to the physical position of the EVCS in relation to the inverter.

The description for 5026 is correct, the description for 3027 is incorrect

izak, 2026-10-06:
You probably mean 1026 and 3827.

closed 2026-10-06 by izak

gb095666, 2026-10-06:
No, the original number were correct, there are two manuals which show Modbus register numbers.

The overall (CCGX) spreadsheet shows registers for the EV charger (and many other devices) which is for an integrated system where there is also solar and battery charging etc, This shows the position of the evcharger as 
0=AC input 1;1=AC output;2=AC input 2

The other (EVCS) spreadsheet shows the registers for the EV charger if it's a standalone install which shows different values for the position of the EVcharger
0 - Inverter Ac Out, 1 - Inverter Ac Input

Note AC out is 1 for an integrated setup and value 0 for standalone

My setup is integrated but was showing the standalone value instead of the integrated value