victronenergy/dbus_modbustcp #50: History.Charged/DischargedEnergy with CAN bus batteries

jens-decker-bruker, 2025-09-16:
These two values are vital to integrate the Victron Multiplus into very famous Home Assistant to report the energy flow from/into the battery and do further analysis. But they are both just zero. I got the information that this is an issue for CAN bus connected batteries, which is the preferred way without real alternatives?

I would really vote for a fix of that. As I can see the energy flow in VRM the numbers should be available somehow also for my system.

mgies, 2025-09-16:
What type of battery are you using?

jens-decker-bruker, 2025-09-16:
Its a DIY battery based on a EEL kit with a Seplos BMS

Thats what it looks like in VRM:

[image]

jens-decker-bruker, 2025-09-16:
I was trying this code: https://communityarchive.victronenergy.com/questions/266290/include-battery-data-in-home-assistant-energy-dash.html fixing the indents and duplicated power_discharge name but didn't succeed. Writing this reply I see, that also the unique_id is the same for both sensors, didn't change that yet. Maybe I will succeed also if I change that in addition ...

izak, 2026-10-07:
Register 301 and 302 holds this information, and it has been in Venus OS since before version 2.00.

If the values are not available, it is most likely because the battery does not publish them. Many CAN-bms batteries don't publish the energy values. On modbus they will then show up as a zero. It would have been better if it was published as 0xFFFF (not supported), but that is hard to change now.

Publishing of the energy values is also supported in the can-bus-bms driver since Venus version 2.40. As long as the battery published message id 0x378, it will work.

Closing this issue as this is not a shortcoming in modbus_tcp.

closed 2026-10-07 by izak

mgies, 2026-10-08:
Yes, the Pylontech BMS and others may not report these Values, yet they are shown in the CerboGX UI and VRM and are thus caculated somehow. 

Why wouldn't it make sense to publish this calculated value through modbus_tcp as well? 

There could be either a logic, regognizing the missing BMS value and then using the calculated, or just expose a new datapoint for the calculated value.

mgies, 2026-10-19:
@izak Please read my last comment and be so kind to reconsider if that could be a solution. After all, VRM shows calulated values, so why shouldn't these calulated values not be made available through modbus as well?

izak, 2026-10-19:
These values are calculated by vrmlogger from all the other energy counters, and are not available on dbus at all, therefore cannot be easily exposed on modbus, at least not yet. If we do add them, they would represent system-wide values rather than specific to CAN-bus batteries. We may in fact do this in the near future.

What you can do right now is to calculate them yourself, exactly as we have to do it.

If you take the energy counters of the solar chargers (register 790), and subtract the energy consumed by the Multi/Quattro (registers 86, 88, and 90), then add whatever came from the AC side (76 and 80), and do this in 15-minute buckets, and assume a 90% conversion rate on the outgoing leg, you will get similar estimates to the ones we make for VRM.

I'm perfectly willing to make a solution, I just don't think that solution is to estimate energy counters for batteries that don't have them. A better solution is to make the values we estimate for use on VRM available.

If we do decide to estimate energy for batteries that don't have them -- I think the Node-Red virtual battery implementation might already do that -- it should also live in the can_bus_bms driver (closed source) and not in dbus_modbustcp.